linumiz/zephyr #1: Using this library with nRF9160DK, compatible with Zephyr

adeel1710, 2020-10-21:
Hi , I am using the liblightmodbus library on a nordic nRF9160DK board that is connected to the external modbus communicated sensor. I had to read the holding register values from the sensors and I am tring to use the :  modbusBuildRequest0304( &mstatus, 3, 1, 0, 16 ); function and use request0304 struct to transmit a modbus frame from the UART. 

Is this the right way to use this library for external communication ? I am using this library for the first time and would appreciate any help in the explanation.

Regards,
Adeel.

parthitce, 2020-10-30:
@adeel1710 liblightmodbus helps in building the modbus RTU data. Sending the data over UART/RS484/RS232, you have use UART driver API provided by Zephyr.

adeel1710, 2020-10-30:
Hi Parthiban,

Yes, I understood most of the part in the build function. 
I am trying to use Zephyr UART library to send out data (basically : uart_fifo_fill() function). But I was confused as to which data field has to be exactly sent over the UART driver. 
I assumed that I could send : mstatus.request.frame[] in order to send the modbus data frame out through UART.

Could you let me know exactly which field builds the complete modbus data that I could send out.

Thanks for the help :)

Regards,
Adeel.

parthitce, 2022-08-15:
@adeel1710 zephyr already got native modbus support. Closing it as won't fix!

closed 2022-08-15 by parthitce