linumiz/zephyr #30: UART PM

parthitce, 2024-11-22:
PM: Power management

Suspend, resume with handling for pinmux states changes, clock disablement or switch low power clock handling. This needs to be done under device power management.

parthitce, 2025-05-08:
[image]

PM_DEVICE_RUNTIME support for UART needs to be done.

testo-jeff, 2025-05-09:
@parthitce  This is the top priority issue right now, it won't block today's release, but we need this feature in next version, thx.

lu-ka-s, 2025-05-09:
The main issue is
> STANDBY stops delayable work timeout

https://github.com/linumiz/zephyr/blob/a4d393ee8944959a7fc85655c7fb8ba122d38de7/drivers/serial/uart_mspm0.c#L217-L222

This timeout will never occur because systick is stopped.
Fixing this has the highest priority.

parthitce, 2025-05-27:
> What is the status of this issue? I have two questions:
> 
>     * When TI designed the UART, they made it work in STANDBY mode. What is their intended way to implement the DMA rx timeout?

Working in STANDBY means waking the SoC from STANDBY and resume it. But w.r.t DMA RX, IMO still we need to make the device busy and not allow to sleep state when there is an active transaction on-going.

> 
>     * When TI designed the "fast clock request", did they provide a convenient way to find which device asserted the request?

I don't see a way to know that. But ISR of respective IP should be the one which will trigger upon resuming the SoC. But I had/having a question, when there is pulse in UART or I2C lines from the other end/host, it won't be waiting for the MSPM0 to fully resume from sleep. So this already misses a transaction?

lu-ka-s, 2025-05-28:
> What is their intended way to implement the DMA rx timeout?

Have you seen the DMA UART implementation in TI's MSPM0_SDK?
I recommend reading `MSPM0_SDK_2_04_00_06/source/ti/drivers/uart/UARTMSPM0.c`. They are using the UART FIFO receive-timeout interrupt.