linumiz/zephyr #32: Clock Control PM

parthitce, 2024-11-22:
- Support for CONFIG_PM_DEVICE with STOP0 & STANDBY0 level corresponding to suspend/resume/on/off needs to be added
- Set sys clock to 32MHz power system down to STOP2 or STANDBY0 mode

PM: Device Power Management

lu-ka-s, 2025-05-05:
Regarding https://github.com/zephyrproject-rtos/zephyr/issues/89031 and https://github.com/zephyrproject-rtos/zephyr/issues/84049

With the proposed deprecation of `PM_DEVICE_SYSTEM_MANAGED` and the elimination of all calls to `pm_device_action_run`, how will the application make use of the clock control power management?

lu-ka-s, 2025-05-05:
Also, currently only resume is implemented:
https://github.com/linumiz/zephyr/blob/391adc2046b19645b6682e17b3b88d33469f623d/drivers/clock_control/clock_control_mspm0.c#L250-L267

bjarki-andreasen, 2025-05-05:
> Regarding [zephyrproject-rtos#89031](https://github.com/zephyrproject-rtos/zephyr/issues/89031) and [zephyrproject-rtos#84049](https://github.com/zephyrproject-rtos/zephyr/issues/84049)
> 
> With the proposed deprecation of `PM_DEVICE_SYSTEM_MANAGED` and the elimination of all calls to `pm_device_action_run`, how will the application make use of the clock control power management?

Probably some device will need the clock to be on to function properly, this device will have a reference to the clock, and will call pm_device_runtime_get() to enable it. This is very similar to calling pm_device_action_run(), but is reference counted and supports PM sleep states and power domains as well :)