linumiz/zephyr #45: Power Management

parthitce, 2024-11-22:
- Peripheral level detailing is mentioned in each IP’s above
- Timer for the scheduler decides the variance between performance and low power modes. Implementing systick backend with timer (drivers/timer) needs to be done to use LFCLK or 32.768 Low frequency accordingly
- SoC level power management needs to be handled
- Features like suspend and resume from RAM is handled in SoC specific way as there isn’t any abstracted PM handling. For example, Nordic handles this well and probably this needs to be inherited here.

lu-ka-s, 2025-05-05:
`DL_SYSCTL_setPowerPolicyRUN0SLEEP0` overwrites MCLK clock source
https://github.com/linumiz/zephyr/blob/391adc2046b19645b6682e17b3b88d33469f623d/soc/ti/mspm0/mspm0l122x_l222x/power.c#L74-L78

The clock tree state needs to be restored correctly

lu-ka-s, 2025-05-08:
> Implementing systick backend with timer (drivers/timer) needs to be done to use LFCLK or 32.768 Low frequency accordingly
Currently systick will slow down in STANDBY0 and stop in STANDBY1, failing to wake the system after the expected number of ticks

parthitce, 2025-05-08:
> > Implementing systick backend with timer (drivers/timer) needs to be done to use LFCLK or 32.768 Low frequency accordingly
> 
> Currently systick will slow down in STANDBY0 and stop in STANDBY1, failing to wake the system after the expected number of ticks

The initial idea was fully different based on the proposal. But with current state of things, systick will not or should not run in STANDBY modes. It should resume only after fast clock request. In the energy analyzer I can see that there is no systick happening at all in STANDBY. Do you see otherwise?

lu-ka-s, 2025-05-08:
My bad, I see now that SYSCTL_MCLKCFG_USELFCLK_ENABLE is only set in SLEEP1.
I misidentified another timer interrupt as SYSTICK.

But I can see a missed opportunity here:

[image]

[image]

The availability of fast clock requests on TIMG0 and TIMG1 could be used to support waking from STANDBY after the number of ticks that the kernel passed to `bool pm_system_suspend(int32_t kernel_ticks)`. This way the application can enter the most efficient power mode with a simple `k_sleep` without having to worry about missing the deadline

lu-ka-s, 2025-05-08:
Right now, the default policy will use the [timings defined here](https://github.com/linumiz/zephyr/commit/45f063856c9bd6e2bc3a37ab4876de784e1cc4b5) to choose STANDBY if the application happens to idle for over 26ms at a time. So calling `k_sleep(K_MSEC(30))` is dangerous because the device will need an external interrupt to ever wake again. This behavior is unexpected IMO

parthitce, 2025-05-09:
> Right now, the default policy will use the [timings defined here](https://github.com/linumiz/zephyr/commit/45f063856c9bd6e2bc3a37ab4876de784e1cc4b5) to choose STANDBY if the application happens to [...]

As stated in the commit message, `min-residency-us` is more of a rough value. But the real ones needs to be decided based on the application and usecase. Otherwise in a nutshell is to define the power policy custom based on the usecase, which can involve factors other then time as well.

lu-ka-s, 2025-05-12:
"Not entering STANDBY" is not solving the problem.

With the current side effects, every timer in the application that is using systick will drift once STANDBY is entered.
I do not see how we should use a custom policy to prevent that.

lu-ka-s, 2025-05-19:
RUN0SLEEP0 should be removed from the list of power states. It is equal to ACTIVE.
https://github.com/linumiz/zephyr/blob/f027eb6a7db6e2e4a0aacb4180f9a51e616e7a98/soc/ti/mspm0/mspm0l122x_l222x/power.c#L10-L15
We should avoid calling `DL_SYSCTL_setPowerPolicyRUN0SLEEP0` as it will overwrite the clock tree.

RUN1SLEEP1, RUN2SLEEP2 and STOP2 will all overwrite the current clock tree. These changes need to be reverted in the exit_post_ops:
https://github.com/linumiz/zephyr/blob/f027eb6a7db6e2e4a0aacb4180f9a51e616e7a98/soc/ti/mspm0/mspm0l122x_l222x/power.c#L75-L79

[image]