linumiz/zephyr #60: MSPM0L1228: Add SoC and dts files

parthitce, 2025-04-02:
L2228 is added with SoC and DTS files based on the launchpad hardware. Add L1228 as well. Two main difference being,
- Flash size is large and organized in 2 banks
- No LCD, this should make no difference at this stage

parthitce, 2025-04-03:
Addressed with https://github.com/linumiz/zephyr/compare/a09219d56d85...669715b450cd

lu-ka-s, 2025-05-16:
https://github.com/linumiz/zephyr/blob/f027eb6a7db6e2e4a0aacb4180f9a51e616e7a98/dts/arm/ti/mspm0l122x_l222x/mspm0l1228.dtsi#L1

does not compile for me, maybe just use this

https://github.com/linumiz/zephyr/blob/f027eb6a7db6e2e4a0aacb4180f9a51e616e7a98/dts/arm/ti/mspm0l122x_l222x/mspm0l2228.dtsi#L1

parthitce, 2025-05-20:
@testo-jeff  Do you like to re-open this issue?

testo-jeff, 2025-05-21:
@parthitce  See lukas's comments, seems it doesn't work but low priority.

testo-jeff, 2025-08-07:
@parthitce  Seems the dts files for MSPM0L1228 still has some issues when we config it like below:

[image]

Use MSPM0L2228 can compile successfully but the MCU model is incorrect.
Use MSPM0L1228 can't compile successfully.

Would you please fix it?

parthitce, 2025-08-11:
> [@parthitce](https://github.com/parthitce) Seems the dts files for MSPM0L1228 still has some issues when we config it like below:
> <img alt="Image" width="206" height="70" src="https://private-user-images.githubusercontent.com/208913967/475348674-621f3f73-90b4-45ca-8053-2d297757ae2c.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQ4ODU4MTgsIm5iZiI6MTc1NDg4NTUxOCwicGF0aCI6Ii8yMDg5MTM5NjcvNDc1MzQ4Njc0LTYyMWYzZjczLTkwYjQtNDVjYS04MDUzLTJkMjk3NzU3YWUyYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODExJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgxMVQwNDExNThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yMDliMWY3NzY4NDlkMDIwMDQ2YjZmZDkxZDI1NDkyYjFmNTUxNmIzNTdmZTc4OGM5MjE1MzhlMDEzYTJkOWM0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.lSCsG7Ist424NGxCRJd6Iuv1VLd-w7ZGNftUkBsNwDg">
> 
> Use MSPM0L2228 can compile successfully but the MCU model is incorrect. Use MSPM0L1228 can't compile successfully.
> 
> Would you please fix it?

@Girinandha-M Can you please take a look at this issue and test it?

parthitce, 2025-08-14:
@testo-jeff Addressed with https://github.com/linumiz/zephyr/pull/218

closed 2025-08-22 by parthitce