linumiz/zephyr #63: MSPM0: JLink issue

parthitce, 2025-04-02:
Segger based flashing script needs to be prepared. Actual issue with flashing and reset needs to be added by @lu-ka-s here

lu-ka-s, 2025-04-02:
[image]

This looks like the device needs a reset before flashing.
I think there are many ways to reset this device, including a SEC-AP mailbox.
Maybe JLink uses the wrong type?

Uniflash and CCS are configured to reset before progam load

[image]

parthitce, 2025-04-02:
https://pastebin.com/raw/pYruj8U4 - Openocd
https://pastebin.com/raw/L02VCNFq - JLink working version
https://pastebin.com/raw/7VFycuwT - JLink failure

Public reference to the error logs.

parthitce, 2025-04-09:
```
****** Error: Timeout while preparing target, RAMCode did not respond in time!
Failed to perform RAMCode-sided Prepare()
Unspecified error -1
```

parthitce, 2025-04-10:
```
J-Link>loadfile /home/parthiban/work/west/mainline/zephyr/build/zephyr/zephyr.hex
'loadfile': Performing implicit reset & halt of MCU.
Memory map 'before startup completion point' is active
ResetTarget() start
DAP initialized successfully.
ResetTarget() end - Took 2.66ms
Device specific reset executed.
Downloading file [/home/parthiban/work/west/mainline/zephyr/build/zephyr/zephyr.hex]...

****** Error: Timeout while preparing target, RAMCode did not respond in time!
Failed to perform RAMCode-sided Prepare()
Unspecified error -1
J-Link>Regs
PC = 00000574, CycleCnt = 00000000
R0 = 200018C8, R1 = FFFFFFFF, R2 = FFFFFFFF, R3 = FFFFFFFF
R4 = FFFFFFFF, R5 = FFFFFFFF, R6 = FFFFFFFF, R7 = FFFFFFFF
R8 = FFFFFFFF, R9 = 400004FC, R10= FFFFFFFF, R11= FFFFFFFF
R12= FFFFFFFF
SP(R13)= 20000898, MSP= 20000898, PSP= FFFFFFFC, R14(LR) = 00002EE3
XPSR = 01000002: APSR = nzcvq, EPSR = 01000000, IPSR = 002 (NMI)
CFBP = 00000000, CONTROL = 00, FAULTMASK = 00, BASEPRI = 00, PRIMASK = 00
FPU regs: FPU not enabled / not implemented on connected CPU.
```

parthitce, 2025-04-10:
Addressed with https://github.com/linumiz/zephyr/commit/0f98723fbc614d0e502ec8775796842f77fd8ac8

closed 2025-04-24 by parthitce