linumiz/zephyr #92: PM: Disable WDT in low power mode

parthitce, 2025-04-15:
The watchdog need to can be turned off before entering low-power mode.

parthitce, 2025-04-22:
```
git diff subsys/task_wdt/
diff --git a/subsys/task_wdt/task_wdt.c b/subsys/task_wdt/task_wdt.c
index 361acc0e156..5f48275e478 100644
--- a/subsys/task_wdt/task_wdt.c
+++ b/subsys/task_wdt/task_wdt.c
@@ -179,7 +179,7 @@ int task_wdt_add(uint32_t reload_period, task_wdt_callback_t callback,
                        if (!hw_wdt_started && hw_wdt_dev) {
                                /* also start fallback hw wdt */
                                wdt_setup(hw_wdt_dev,
-                                       WDT_OPT_PAUSE_HALTED_BY_DBG);
+                                       WDT_OPT_PAUSE_IN_SLEEP);
                                hw_wdt_started = true;
                        }
 #endif
```

testo-jeff, 2025-04-24:
This is verified and use task_wdt, is ok now.

closed 2025-04-24 by parthitce