linumiz/zephyr #94: PWM: capture

parthitce, 2025-04-15:
Using the example you provided, I tested the interface function for zephyr pwm capture and have the following problem:

  1.  If I use the pwm_capture_cycles function to get the pwm capture values for a single capture the values captured are different and incorrect each time.
  2.  If I use the pwm_configure_capture interface, I can only block and wait for 500ms to obtain the captured value. If I use a semaphore to wait for the captured value, the value obtained is different and incorrect each time.
  3.  If the capture input index is set to 1 (ti,cc-index = <1>; for example, the pwm capture pin is tima0c1_pa9), the obtained period and pulse width are equal, and the duty cycle is fixed at 100%.

bzhou88, 2025-04-25:
> 1）The current PWM capture function always displays a timeout

The latest code compiled from GitHub is also experiencing timeouts on the development board. I reverted to the previous code, and the PWM capture functionality is working.   @ssekar15 @parthitce 

[image]

> 2）pwm captures pwm wave with frequency 20KHz

I noticed that the resolution in the CCR PWM capture configuration does not differ significantly, So, is there an issue on our part?@ssekar15 @parthitce 
[image]

>  3) PWM output cannot be output at the same time in the same timer
The image below shows that there are two outputs

[image]

ssekar15, 2025-04-25:
> The latest code compiled from GitHub is also experiencing timeouts on the development board. I reverted to the previous code, and the PWM capture functionality is working. [@ssekar15](https://github.com/ssekar15) [@parthitce](https://github.com/parthitce)

@bzhou88 I figured out that you are using CC index 1 which has regression, now fixed please pull the latest

> I noticed that the resolution in the CCR PWM capture configuration does not differ significantly, So, is there an issue on our part?

Above statement is not clear for me. I could observe the same issue on CCS too when input PWM frequency is high like 10KHz or 20KHz.. The resolution factor has impact, I was investigating yesterday in that direction of adjusting capture time & resolution but same result..

testo-jeff, 2025-04-29:
Here is ssekar's analysis, the hardware register values are inconsistent with higher frequency of input which is same behavior on the CCS. Need a clarification with TI support.

[Jeff]
Have you already reported this topic to TI support? Any feedback?

testo-jeff, 2025-05-08:
Verified and ok

closed 2025-05-08 by parthitce