linumiz/zephyr #96: ADC: ADS131 PM

parthitce, 2025-04-17:
[image]
Address the power management handling for the ADC ADS131

lu-ka-s, 2025-04-25:
[image]
The clock output is enabled after init. Calling the pm action will disable it correctly

testo-jeff, 2025-04-25:
@parthitce This one seems need to reopen it as Lukas found some issues.

lu-ka-s, 2025-04-25:
```
/**
 * @brief Enable device runtime PM
 *
 * This function will enable runtime PM on the given device. If the device is
 * in #PM_DEVICE_STATE_ACTIVE state, the device will be suspended.
 *
 * @funcprops \pre_kernel_ok
 *
 * @param dev Device instance.
 *
 * @retval 0 If the device runtime PM is enabled successfully.
 * @retval -EBUSY If device is busy.
 * @retval -ENOTSUP If the device does not support PM.
 * @retval -errno Other negative errno, result of suspending the device.
 *
 * @see pm_device_init_suspended()
 */
int pm_device_runtime_enable(const struct device *dev);
```

Should the driver call this during init or is it better to handle this in the application?

testo-jeff, 2025-04-28:
Verified and ok.

closed 2025-04-28 by parthitce