linumiz/zephyr #97: I2C: PM: Waktup source

parthitce, 2025-04-22:
Details needs to be added by @lu-ka-s 
With current understanding, the device needs to wake from STOP mode to RUN when there is I2C gets SCL from the controller when acting as target.

lu-ka-s, 2025-04-25:
Is it possible to wake the mcu this way when the clock is sourced from an external oscillator?

parthitce, 2025-04-25:
> Is it possible to wake the mcu this way when the clock is sourced from an external oscillator?

Yes it should be. AFA I2C is IP is concerned, it should use the provided BUS clocks (MCLK / ULPCLK / ...). Source of which could be anything.

parthitce, 2025-04-27:
https://github.com/linumiz/zephyr/commit/87317d08e95e2d21ac935fc0f41c4b2fba755cf4 as discussed in chats, enabled the wake from low power mode.

lu-ka-s, 2025-04-29:
I am currently testing the wake up behavior of the UART. All wake up logic is related to "fast clock requests".

Test setup:
python client sending serial commands to MSPM0L1228
a custom `pm_policy_next_state` always returns the active state mapping to RUN0SLEEP0
Oscilloscope connected to SPI0_CK and CLK_OUT (sourced from ULPCLK), triggered on SPI0_CK.

Test procedure:
- power on the MSPM0L1228, SYSOSC@32M is fine and it has never entered STANDBY
- send command (read spi adc)
- receive result after a few milliseconds (see image 1)
- send command (force standby), the MSPM0L1228 enters STANDBY (35uA power draw)
- send command (read spi adc), the MSPM0 wakes up
- receive result after **6 seconds** (see image 2 and 3),

I can see that the MSPM0L1228 successfully receives the command an is capable of executing code. But it looks like it is going back to STANDBY for some reason. The custom policy will always return the active run state info, so why is it going back to STANDBY?

Image 1: normal spi communication
[image]

Image 2: spi command executed after waking up from standby (UART fast clock request)
[image]

Image 3: a long time later, the clock output shows the ULPCLK was sourced from 32k LFCLK
[image]

lu-ka-s, 2025-04-29:
I found the issue. This code can not be reached:
https://github.com/linumiz/zephyr/blob/7836e1eacd51c2267c34b103ac369d91028c83ad/soc/ti/mspm0/mspm0l122x_l222x/power.c#L51-L56
meaning that this is never executed:
https://github.com/linumiz/zephyr/blob/7836e1eacd51c2267c34b103ac369d91028c83ad/soc/ti/mspm0/mspm0l122x_l222x/power.c#L13

The Zephyr idle task ignores all `PM_STATE_ACTIVE` states:
https://github.com/linumiz/zephyr/blob/7836e1eacd51c2267c34b103ac369d91028c83ad/subsys/pm/pm.c#L159-L179

The solution is to call `DL_SYSCTL_setPowerPolicyRUN0SLEEP0` from the state exit function to apply the run time clocks during resume:
https://github.com/linumiz/zephyr/blob/7836e1eacd51c2267c34b103ac369d91028c83ad/soc/ti/mspm0/mspm0l122x_l222x/power.c#L75-L78

The idle task calls it here:
https://github.com/linumiz/zephyr/blob/7836e1eacd51c2267c34b103ac369d91028c83ad/subsys/pm/pm.c#L119

parthitce, 2025-04-29:
@lu-ka-s please re-run the SPI wake and UART using the latest changes and write back.
Here are changes for clock PM: https://github.com/linumiz/zephyr/commit/cc2b88d90196d35905943b17a3f0d94d6d0be819
and https://github.com/linumiz/zephyr/compare/7836e1eacd51...b1114ad021e5 for the overall PM states

lu-ka-s, 2025-04-30:
My tests are passing now. It looks good.

lu-ka-s, 2025-05-12:
We have observed soft resets if the device is repeatedly woken from STANDBY0.

Setup:
- pm policy that always returns a fixed mode
- configure I2c as peripheral
- set power mode standby0

Now repeatedly scan the i2c bus with a different board. The MSPM0L1228 will wake to ACK and go back to sleep a few times, and then suddenly resets

lu-ka-s, 2025-05-13:
In our test there is only one task_wdt channel configured with a timeout of 2 seconds.
The feeding thread is just a loop that sleeps for 1 second.
Both the threads `k_sleep` and the task_wdts 'k_timer' use the systick timer, so there can be no relative drift.
During STANDBY0 systick is stopped. The only explanation for the triggered task_wdt is that the feeding thread has been blocked for over 1s by interrupt service routines and other threads.

Our test is using UART at 115200 baud to send the I2C bus scan command to the device and the device will answer with a list of addresses that acknowledged the scan message.

I think we have managed to overload the system with ISRs.
Is there a way to profile a zephyr application? I am interested in the % time that was spent in ISRs to verify my thesis.

parthitce, 2025-08-14:
@d-philpot @JFMSP This is the internal open issue to track the systick wake issue. README to reproduce the issue is here: https://github.com/linumiz/zephyr/blob/dev/ti/mspm0-systick/samples/boards/ti/mspm0/README.md