linumiz/zephyr #100: Timer only support ch0 pwm, can't support ch1 pwm

testo-jeff, 2025-04-25:
I set up two PWM outputs on the same timer, but only one of them is working. When I use CCS, I can get both PWM outputs to work at the same time.

lu-ka-s, 2025-04-25:
Possibly caused by this line
https://github.com/linumiz/zephyr/blob/ff14dd2d06fd38bc5a232819361daaa692deb8a4/drivers/pwm/pwm_mspm0.c#L87

bzhou88, 2025-04-28:
I measured the output of a 5Hz PWM wave and found that the duty cycle output is incorrect. @ssekar15 
I have achieved an output PWM frequency of 5Hz with a duty cycle of 80%; however, the oscilloscope displays a duty cycle of 15%.

[image]

testo-jeff, 2025-04-29:
Verified and ok

closed 2025-04-29 by parthitce