linumiz/zephyr #101: Counter overflow case

testo-jeff, 2025-04-25:
If the PWM period I need to capture exceeds the maximum time of the timer, an overflow interrupt will occur, and the captured result will also need to include the time before the overflow.

As shown in the figure below:

[image]

saravanan said:
The PWM capture signal could not spread over multiple counter period.
The maximum capture period is caped to counter maximum value (dts ti,period = <62499>), CCS does the same.
I will add a fix to avoid notification on overflow case.

bzhou88, 2025-04-25:
I have noticed that there is a submission regarding this issue; however, after implementing the code from this submission, the PWM capture functionality does not work and consistently prompts a timeout.

testo-jeff, 2025-04-28:
For  the counter overflow issue, here is an example for better understanding.

Use 32MHz as clock source, 16bit counter, so the timeout is 1/32M*65535 = 2.0486ms
Use 5Hz as input source, 1 circle is 200ms,  which means meansure 1 cycle need around 100 times overflow( 200/2.0486)

testo-jeff, 2025-05-08:
Verified and ok

closed 2025-05-08 by parthitce