linumiz/zephyr #107: Timer: system timer support

parthitce, 2025-05-20:
With counter handling the system timer implementation, there are two issues which needs to be addressed
- systick with counter alarm support, delta is incorrect and waking from the standby1 is isconsistent
- With STANDBY1, the tick moves to the future time. Can be tested with https://github.com/lu-ka-s/systick_standby_drift/blob/lu/systick_glitch/samples/boards/ti/mspm0/standby-systick/src/main.c

parthitce, 2025-05-20:
Addressed with https://github.com/linumiz/zephyr/commit/3d40414f763dcf20d6bea3e2219a7167d3c23ef7
@lu-ka-s could you please test this changes?

lu-ka-s, 2025-05-22:
The 32.768 kHz clock source and the 255 prescaler result in a counting frequency of 128 Hz. If the kernel tries to enter a power state for less than 1/128 seconds the `counter_alarm_cfg` will have a value of 0 ticks. Setting an alarm for 0 ticks will not work.

There is the `set_guard_period` counter api for this purpose but `cortex_m_systick.c` does not use it.
There is another flaw right next to it. The return value of `counter_set_channel_alarm` is discarded, resulting in a missing alarm if the kernel wants to sleep longer than the counters top value...

https://github.com/linumiz/zephyr/blob/fb4bb5d7688f5b232023ea07308edfe734f41b7e/drivers/timer/cortex_m_systick.c#L249-L265

testo-jeff, 2025-05-22:
Hi @parthitce @ssekar15 , just aligned with Lukas, the systick fixing by idle_timer may not a good option per Lukas test, because it caused other issues(automatically wakeup in standby mode), would you please try to find out the root cause of systick jumping issue with other possible solution?

@lu-ka-s Would you give more information here? Thx.